BotoX/xiaomi-m365-compatible-bms #1: Xiaomi bms vs alternative

lednicazar, 2018-12-22:
Hello BotoX,
I have a faulty battery with some dead cells due to a faulty bms.
I want to rebuild the 10S3P original battery without change the cell config, just rebuilt it. But I have the doubt choosing the BMS replacement:
What do you think, should be better this bms compared with the original one? (I've read a lot of faulty xiaomi bms...)
If I pick this one should I take the 10S li-ion 30A BMS on the web page or with the 20A should be enough?

Thanks!

videofunk, 2018-12-22:
I m not BotoX.bz. His fb is @david bodrużîć... BMS need 35Ampere.

lednicazar, 2018-12-22:
Hi Videofunk!, Thanks for the answer. I'm asking that because BotoX tell that versions above 30A will not fit in the housing. 
Why is needed above 30A? Due to the motor?

BotoX, 2018-12-22:
The versions above 30A have both sides of the PCB populated with mosfets and are thicker.
20A or 30A should be fine, I just went with 30A to be sure (just a few more bucks for better mosfets)
It'll be easier to use the original xiaomi BMS because you can just plug it in, here you'll have to solder the balance wires.

lednicazar, 2018-12-22:
Thanks BotoX,
but your feeling  is that is more stable that bms ? Or should be at the same range?
I know that will be more complex, but if that bms will be more reliable maybe makes sense the extra effort, no?
And in terms of cost both are at the range of 30/40€.

BotoX, 2018-12-23:
It's probably more reliable, yeah.
And if you think it's the worth the trouble then go for it.
I mainly did this so I can do 12S with my scooter.

lednicazar, 2018-12-23:
Purchased the 30A. I will update the post with news as soon as I tested it.

lednicazar, 2018-12-27:
Purchased de bms and 30 Samsung INR18650-35E, with the idea to rebuild completely the battery upgrading the capacity up to 10.200mAh (should be more than 30% of improvement, right?).
What's your opinion of that cells? (Purchased in nkon, so should be official ones).
About building the battery, do you use nickel strips or the ones that are with shape to connect several cells?

BotoX, 2018-12-27:
The cells are fine, a little expensive but good performance.
Nkon is a reputable source.

But I don't understand why you don't just buy 48 cells (could use cheaper ones too then like 29E) and build a 12S4P battery when you're already building a new battery?

I spotwelded my battery with 0.2x7mm pure nickel strip.

lednicazar, 2018-12-27:
Mainly I did that because was a m365 purchased with the bat broken, without knowing how many dead cells are in the battery...
I expected less cells broken, so I purchased the 10S BMS, and yesterday finally i was able to test the battery... And only 8 cells above 3v, and 14 between 1-2v, and 8 close to 0v....
So I decided to buy 30 new and bigger(instead of replace the defective ones) to use the BMS purchased.
At the end with 30% more range I think I will be able to do for sure close to 35km, that should be enough...
But thinking all slowly makes sense the other option... I will check with this and in the future I can also build the other.

I will try to explain here all the steps!
With a imax b6 I will be able to repair the cells with 1-2v??