BotoX/xiaomi-m365-compatible-bms #5: Benefits of upgrade?

tbone-rd, 2019-03-16:
This project looks very intersting!

- First, my config and problem:

I have only flashed the customized firmware 1.4 (Motorconstant 34383 / 750W ) for now.
With full battery and 100kg going up the hill, full throttle, i get 19-20amps on dash display.
I hear a "Klack ... Klack", and torque feels pulsated.
When battery is not full, it does not happen, torque just sags.

Does the battery not deliver enough current or is the motor to weak?

- BMS upgrade

I'm curious what the benefit of this BMS upgrade is.
More range because of higher battery capacity, ok.
But for what is the voltage upgrade..only higher top speeds? Lower amps with same torque?
Would higher voltage "protect" the motor from to much amps when going uphi

BotoX, 2019-03-16:
higher voltage = higher speed for longer, more torque, more power
more/better cells = more capacity/range, less voltage sag, higher cycle life

The original battery is a waste for this scooter, it can do much better.
The next upgrade I'm looking forward to is a better motor, not sure where I'll find one though ^^

tbone-rd, 2019-03-17:
Do you have an idea about my "Klack .. Klack" Problem?
Rather Motor or Battery weakness?

Yes, i was also looking around for better motor.
But looks hard to find one that fits, maybe the motor for the pro (but only 300w) will be an option
https://youtu.be/nMg0NjCaFfk

BotoX, 2019-03-18:
No clue about the klack klack issue, try posting a video in a telegram group, maybe somebody had this issue and solved it.
Could be a broken hall sensor maybe.