BotoX/xiaomi-m365-compatible-bms #6: Noob question

lukasd1, 2019-03-17:
Do somebody know how to upload this code with USBTinyISP programmer ?

BotoX, 2019-03-17:
https://docs.platformio.org/en/latest/platforms/atmelavr.html

lukasd1, 2019-03-17:
I tried this and i think I'll too stupid for platformio or it is piece of big shit.... , there is no problem with normal arduino and similar but this shit is a different level of abstraction.....

lukasd1, 2019-03-17:
ok ...... i found solution :P

BotoX, 2019-03-18:
You can always get the hex file that platformio makes and flash it with any other software you are familiar with.

lukasd1, 2019-03-20:
OK , I've done everything right with your instruction , now BMS is working (voltage on output is ok ) but there is no communication with ESC (i connected the wires like you write but it's not working :/)