BotoX/xiaomi-m365-compatible-bms #8: UVP Doesn't recover

ryci-us, 2019-03-28:
Hi,

I have custom made battery in 12s3p configuration. Today when i was going home i totally discharged my scooter battery and UVP protection engaged. I came back home and connected charger - i was not charging. I let it sit for 2 hours and it didn't started to charge. So i removed cover, measured voltage between B+ ant B- ant it was 39,6V. So i disconnected/connected balancing connector and it started to work. Looks like that UVP doesn't automatically recover from fault. Can you fix it?

After unplugging/plugging balancing wires i also noticed that full charge/ charge count counters was also reset to 0. 

Bye the way i would like to thank you for great project :)

BotoX, 2019-03-28:
By unplugging / replugging the balance connector your reset the MCU, thus the data got lost and also the issue you were having was fixed.
It would've been interesting which exact error condition was triggered so that the charging port was disabled.
If this happens again please paste the full output of debug_print()

ryci-us, 2019-03-28:
Ok, i will do that. At this time i had a mind to check what is going, but i left my USB-TTL adapter at work :(

About counters: maybe i would be better idea to write these counters to eeprom each time charger is plugged in? Atmel states that atmega328 eeprom endurance is 100,000 write cycles, so bms can outlive a lot of battery packs.

BotoX, 2019-03-28:
I write the data every time the cycle counter is increased.

BotoX, 2019-05-13:
Could be fixed by the latest commit.

ryci-us, 2019-05-28:
I didn't updated firmware, today i had same issue again.
This time again i didn't had USB-TTL adapter, but this time with M365 Battery inspected all parameters, and i think i found what protection i have triggered, it was Over Temperature, because temperature was 44 and 46C. Then battery dropped temperature to 24/24C charging was not enabled.

BotoX, 2019-05-28:
I'd suggest you to flash to the latest firmware, I fixed a few things that might cause this.
If it still happens then I'll investigate this further, never happened to me but it's not that hot here.