BotoX/xiaomi-m365-compatible-bms #10: 13s BMS?

FalloutBe, 2019-04-24:
Hello,

I plan to build a 12s battery pack, but use a 13s BMS, for possible future expansion if people find out how to not blow up other components in the scooter.
Is this a good idea? Or will the firmware not work with the 13s BMS version?

Also, can the 13s BMS operate with only 12s connected?

Another question: will the bluetooth module of the scooter also give access to the app from the BMS manufacturer, as shown on their website?

Kind regards,

BotoX, 2019-04-24:
The firmware doesn't care how many cells your battery has.
The TI BQ76940 chip can be reconfigured according to datasheet: https://cloud.botox.bz/s/JZD5gyQABipz2jg
The original BMS app will not work, you're changing the firmware of the BMS.
However the M365 apps will work, like M365 Tools or M365 Battery and Mi Home, Ninebot, etc.

FalloutBe, 2019-04-25:
Thanks a lot for explaining, and for the fast reply!

I went ahead and placed an order for the 15s 30A version, and 6 extra cells (same ones as in the original battery pack) to convert it into a 12s3p pack ;)
I will probably never use more than 12s for this scooter, but the price for the 15s BMS was not much more than the 12s version.

I plan to put the 6 extra cells in the front bar.
I'm wondering why nobody has done that yet. (as far as I know).
They should just fit vertically between the two holes where the brake cable goes in and out. The tube should also be wide enough to allow for that cable to pass next to the batteries.
This should also put a little more weight on the front wheel to allow for a bit more traction when accelerating.
*wish me luck* :P

I see you defined MAX_NUMBER_OF_THERMISTORS 3, since the bq76940 supports 3 thermistors,
but the 15s BMS module seems to have only 2 thermistor connections. (as far as I can see from the pictures, since I don't have the module yet)
I will have 1 original thermistor and 2 new ones which come with the BMS, so if I want to use all 3 of them, is it easy to solder the original one somewhere? Have you done that before?
Do you know if the original thermistor is also 10k?

Btw, your work is very much appreciated. I imagine you put a lot of time into this project.

BotoX, 2019-04-25:
The BMS only has two connectors for the thermistors. The first thermistor pin is also the boot pin, so it requires special care, gotta try if you can make it work.
The thermistor type used is in the source code.

shajek, 2020-04-10:
Hello, i have 13s pack. For now i have BMS password error, i will try balance cells and I see... but .. m365 tools show only 12cells. Is possible to show all 13cells ? thanks

FalloutBe, 2020-04-10:
Did you connect all the bms wires exactly like the table in the manufacturer's datasheet instructed?Even if your cells are not balanced yet, all 13 cells should show up.

shajek, 2020-04-10:
hello, thank you for reply

Yes i have correct connection. I shorted VC14-VC13 and VC9-VC8 like datasheet say.

shajek, 2020-04-10:
hello, this is log from configtool

```
 debug_print()
b'55aa0322fa0500dbfe'
>>> 0x00 SYS_STAT:  00000000
0x01 CELLBAL1:  00000000
0x04 SYS_CTRL1: 00011000
0x05 SYS_CTRL2: 01000011
0x06 PROTECT1:  00011101
0x07 PROTECT2:  01111001
0x08 PROTECT3   00010000
0x09 OV_TRIP:   10110000
0x0A _TRIP:   11001001
0x0B CC_CFG:    00011001
0x32 CC_HI_LO:  0
0x2A BAT_HI_LO: 31280
ADCGAIN:        377
ADCOFFSET:      47
ER ERROR:     0

uptime: 105
Battery voltage: 47781 (31280)
Battery current: 0 (0)
SOC: 100.00
Temperature: 24.70 24.10
Balancing status: 0
Cell voltages (15):
3553 (9301), 3699 (9688), 3711 (9720), 3633 (9514), 3683 (9645), 3672 (9616), 3720 (9745), 3700 (9691), 47 (0), 3706 (9707), 3680 (9638), 3668 (9605), 3650 (9559), 47 (2), 3700 (9690)
Cell V: Min: 3553 | Avg: 3675 | Max: 3720 | Delta: 167
maxVoltage: 4778
maxDischargeCurrent: 0
maxChargeCurrent: 0
XREADY errors: 0
ALERT errors: 0
P errors: 0
OVP errors: 0
SCD errors: 0
OCD errors: 0

DISCHG TEMP errors: 0
CHG TEMP errors: 0
CHG OCD errors: 0
```

BotoX, 2020-04-10:
BMS Password error is when cell voltage difference is >100mV.
Wait for the BMS to balance the cells

shajek, 2020-04-10:
i thought, as i say above... Thank you, i will charge pack. But what is problem that m365 tools show me only 12cells ? Thank you

BotoX, 2020-04-11:
Well in debug_print there are 13 cells, try m365 battery maybe? I only tried m365 tools with 12S

JakubRybakowski, 2020-04-28:
> Btw. anything above 12S needs modded hardware (cut trace) or else U4 on the controller will blow up.

Can you tell what trace we need cut? Or what else to do?

slazzy98, 2020-04-29:
Faced an issue today with running a moded software with a higher voltage limit.
Created myself a cfw, and then uploaded it via downg and scooter got bricked couldn't check version anymore via down as it cancelled due to retry (it would connect via Bluetooth still though).
Also the motor would not work either.
So I've tried to flash directly via stlink and still it didn't work. Ive tried 60,then 58, 56,52volts and the thing would brick the controller so it would not be accessible via Bluetooth anymore (pro)
But with the non pro had a luck to go up to 56, and above that it would crash and motor wouldn't work.
Obvsl I'm talking about changing the hard limit of voltage. The voltage in the circuit was 42v all the time no matter of the limit.

C0br4, 2020-05-06:
@BotoX Sorry I don't want to hijack this issue but could you elaborate where the trace has to be cut? I've found someone doing this to a ES2 but the m365 controller looks quite a bit different.

BotoX, 2020-05-06:
> @BotoX Sorry I don't want to hijack this issue but could you elaborate where the trace has to be cut? I've found someone doing this to a ES2 but the m365 controller looks quite a bit different.

https://t.me/XiaomiM365Hacking/60645

shajek, 2020-05-06:
Hello, if you look to DS, U4 is capable handling voltage from 13s. There is no problem to cut, but problem with 13s configuration is with breaking/recup. Accelerating is OK. I am working on solution to reduce voltage in recup and workaround OVP.

And yes, m365 does not show 13cells .. but another app m365 battery show it correctly.

Migu5499, 2020-06-02:
> Hello,
> 
> I plan to build a 12s battery pack, but use a 13s or even 15s BMS, for possible future expansion if people find out how to not blow up other components in the scooter.
> Is this a good idea? Or will the firmware not work with the 13s BMS version?
> 
> Also, can the 13s BMS operate with only 12s connected?
> 
> Another question: will the bluetooth module of the scooter also give access to the app from the BMS manufacturer, as shown on their website?
> 
> Kind regards,
HI, did you finish with 15s bms, i bought it, and idk how to procced i need some help

NetBUG, 2020-07-20:
I have 13S system based on 15S BMS. There are two of cell balancing inputs shorted, resulting in the following layout (before balancing): `Cell voltages (13 / 15):
3769 (9863), 3768 (9862), 3774 (9878), 3787 (9912), 3780 (9894), 3779 (9891), 3777 (9886), 3722 (9739), 41 (0), 3763 (9847), 3723 (9741), 3787 (9912), 3752 (9818), 41 (0), 3781 (9896)`
I think 13S BMS with 12 cells will be fine, as long as the missing balancing input is shorted.
Don't forget to set up all voltages correctly.

I had to cut a trace on ESC between the 1000uF capacitor and the input, and to supply 42V to the capacitor (ESC v1.4).

P.S.  Many thanks to @BotoX, it works finally! :) Even with 13S.