BotoX/xiaomi-m365-compatible-bms #11: capacity configuration with additional BMS and battery pack

AdamWeglarz, 2019-05-02:
Hi,

I will use 13s3p with original battery pack with addtional 3x3 cells inside pipe managed by this BMS and in addition I will connect 13s3p with Sanyo with additional BMS, This additional battery pack will be connected to + and - of original battery (so it will be charged through this BMS, but balanced and protected by second BMS). I should state capacity of all cells or only one managed by this BMS?

Adam

BotoX, 2019-05-02:
tough one.
if you connect it in parallel to B- then you should use the total capacity
but then the safety of the BMS is gone because the other battery can charge this one

If you connect it in parallel to C- then you have to use only the capacity of the internal battery pack and you have the protection of the BMS but it might be a bit weird, need to remove charge mode from scooter.

AdamWeglarz, 2019-05-02:
I do not think that there is any risk of overcharge of "original battery" as supplementary battery will be charged through this BMS. So not possible to overcharge this extra battery.

BotoX, 2019-05-02:
Yeah there is the danger of overcharging single cells if the battery is unbalanced.
But if you connect the other battery through that BMSs C- you can fix that.

lucidnx, 2019-08-24:
Hello, I am wondering if ESC will survive 13s. It's 54V fully charged.. with 72V there is need to cut trace and solder mosfets with capacitor but with this BMS it's not recommended...

BotoX, 2019-08-24:
At 13S the ESC will die when you use the brake.
12S is the maximum and I only charge mine to 4.1V too.

lucidnx, 2019-08-24:
there is no way to do it? maybe do modifications to ESC? when you have 2 stock batteries in serial connection (72V) there is recommended to cut trace from battery and solder mosfets + capacitor

BotoX, 2019-08-25:
Yeah if you modify it "cut the trace" and change the capacitor for higher voltage, probably change the mosfets too and add a few extra capacitors at the bottom then you can run up to 80V I'd say