BotoX/xiaomi-m365-compatible-bms #12: Сlock generator

Elsta, 2019-06-02:
Hello BotoX!  
Thank you for the work done and interesting project!  Will this project work correctly if you connect an external oscillator 16Mhz to the controller and change the fusebit without changing the m365 firmware?

BotoX, 2019-06-02:
Yes it will if you change this firmware to use 115200 baud and change F_CPU to 16M.
Also need a different bootloader bin: https://t.me/XiaomiM365Hacking/39862

IonicEV, 2019-12-24:
> Yes it will if you change this firmware to use 115200 baud and change F_CPU to 16M.
> Also need a different bootloader bin: https://t.me/XiaomiM365Hacking/39862

I found later posts from "White Power" claiming it did not worked out as
"avrdude : verification error: content mismatch at byte 0x7e00"

Was it the reason you rolled back the 115200bps bootloader?

Elsta, 2019-12-25:
It didn’t work for me at 115200 baud.  I used both quartz and a generator
with and without a bootloader.

Best regargs!

BotoX, 2019-12-25:
The quarz on the V1.6 BMS is 8MHz.

Elsta, 2019-12-25:
I don’t quite understand what you want to know from me.  I tried to run BMS
v1.5 at a frequency of 16 MHz with a voltage of m328p 3.3v.  Did not work
out.  With an external 8 Mhz quartz, everything worked, but at a low speed
not corresponding to 115200 baud.

Best regargs!

IonicEV, 2019-12-25:
> The quarz on the V1.6 BMS is 8MHz.

Thanks! I am looking forward to program the BMS soon.

IonicEV, 2019-12-25:
@Elsta Sorry for the confusion, I was referring to BotoX commit. Thanks for your clarification too.

BotoX, 2019-12-26:
You can only run this BMS with 115200 baud if you use a 16 MHz quarz OR calibrate it with OSCCAL so that you can do 115200 baud with 8 MHz.
See: https://codebender.cc/sketch:186647#OSCCAL.ino

Elsta, 2019-12-26:
thanks, I will try to play with osccal

Best regargs!