BotoX/xiaomi-m365-compatible-bms #13: Choosing power supply

alexeypetrenko, 2019-08-12:
Hi,
Once again, thanks for your awesome work!

I've got a question regarding a PSU:

I'd like to build 12S4P just like you did. `Mean Well HLG-240H-48A` charger is nominally rated at 48 volt. I thought 12S batteries need 12*4.2 (50.4) volts to charge. From datasheet I see that it is possible to adjust output voltage of a PSU. Did you use it?
Does it mean that I should look for a CC/CV PSU which can be adjusted to 50.4 V?

Another small question: Should I use thicker wires for charging. Are existing ones good for 5 A?

BotoX, 2019-08-12:
It's a adjustable CC/CV power supply, you can set the max current and voltage according to the datasheet (and also a bit beyond that).
I charge with 4A (1A per cell) and to 4.1V per cell for better cycle life.

22 AWG is sufficient for the charging wire, don't use the original connector.
I've exposed an XT60 on my scooter for charging, it's where the brake cable leaves the scooter.
Much easier to charge it like this, don't need to bend down and fiddle with the stupid connector, also much quicker to disconnect.

alexeypetrenko, 2019-08-12:
Thanks for the info.
I've added this information in README in https://github.com/BotoX/xiaomi-m365-compatible-bms/pull/14

closed 2019-08-12 by alexeypetrenko