BotoX/xiaomi-m365-compatible-bms #15: SP15S020 BMS

abcheng, 2019-09-03:
Hi,

So I thought I ordered the same BMS as what you linked to on AliExpress, but it looks like a received a SP15S020 BMS instead. 

Any experiences with this board when you were doing research for a BMS to use for the M365?

It looks it uses a stm8s207 instead of a ATMega328p MCU, and has a  TI BQ76940 IC on it.

BotoX, 2019-09-04:
This firmware is not gonna work on it if it's an STM8.
There is Arduino for STM8 however: https://github.com/stm32duino/Arduino_Core_STM8
So this code could be ported to it, however I don't really feel like doing that :v
Though the most important thing is finding which pins of the MCU do what: eg. I2C for the TI BQ chip and UART and then changing that in the code
The sleep code and powersaving stuff can be ripped out and added back later

abcheng, 2019-09-06:
Yeah, I noticed that the code won't work between Arduino and STM8, but was just wondering if you happen to have any experience with STM8 or coding with it.

That link you sent me looks interesting. I'll take a look and see how much work it would be to port the code over.

I saw a data sheet for the MCU, so hypothetically it shouldn't be too difficult to figure out how it's communicating with the Ti chip.

alexeypetrenko, 2019-09-10:
I've just received my BMS and was shocked to see not the one in the pictures. As you guessed, I also got SP15S020.

I'll leave a link here, so other people would not fall do the same mistake.
Do not buy from this seller :
https://s.click.aliexpress.com/e/D7yaPIhS

uzi-b, 2019-09-16:
hi guys,
so i was thinking of an ultra easy fix.
is it possible to make the M365 controller to ignore the input from the BMS ?(so that it would work with no BMS connected)
(in this way we wont get all the battery status on the app but for me its not that important)

alexeypetrenko, 2019-09-27:
Not really. As far as I know m365 will go in a limp mode if there is no proper messages from BMS.

You can use a separate microcontroller to send fake status messages. I do not think it is worth the hustle. 

IMHO it is better to buy the right BMS.

uzi-b, 2019-09-27:
Thank you Alex,
So my micro controller programming skills are pretty much none existent, so even if i had the right BMS i don't know if i would have been able to program it.
I found some Russian shop in Moscow that ship's a BMS emulator for "just" 50 EUR so i will probably buy it from them.

BotoX, 2019-09-27:
you can make that yourself for 2€ too lol

Passerby77, 2019-11-02:
Hi, Please i need help, i need flash BMS SP15S020.hex  
My BMS Stop Working Because i made a flash of another bms, For This BMS Stop Working,
So i need some one help me.

nassuk, 2020-03-03:
> Thank you Alex,
> So my micro controller programming skills are pretty much none existent, so even if i had the right BMS i don't know if i would have been able to program it.
> I found some Russian shop in Moscow that ship's a BMS emulator for "just" 50 EUR so i will probably buy it from them.

Hi, made the same mistake!, any update? Could you please share the Russian seller link?

rbobey1989, 2020-07-05:
Hello BotoX, excellent your work very documented and very well achieved, I wanted to ask some questions about the bms above 15s, up to 20s of this manufacturer, I have read in previous comments and you say that they are stm8 also I would like to know if they are based on the same IC IT for monitoring and balancing cells, if so and you can corroborate the information I would appreciate it, since after studying something your bms code I think that if so I can adapt it and it should not be so complicated greetings I wait for an answer