BotoX/xiaomi-m365-compatible-bms #16: Update code for Ninebot ESx

Neo003, 2019-09-05:
Hello BotoX,
I try to update the bms code in order to work with Ninebot ES2/4 firmware.
As I understand, the main changes will be the NinebotMessage struct, the M365BMS struct and the M365BMS_RADDR / M365MS_WADDR constants. Is that right ?

Thanks for your help.

(suorry if I'm not on the good place)

BotoX, 2019-09-05:
Dump the communication between the ESC and BMS on your scooter and see if the addr is different. The structures shoukd be the same

scooterhacking, 2019-09-06:
https://github.com/etransport/ninebot-docs/wiki/protocol
Xiaomi only has bAddr whereas Ninebot has bSrcAddr & bDstAddr
Request from ESC to BMS:
bSrcAddr = 0x20 bDstAddr = 0x22/0x23
Reply from BMS to ESC:
bSrcAddr = 0x22/0x23 bDstAddr = 0x20
Structure differs, Ninebot pre is 5A A5. You can check other differences in the link above.

Neo003, 2019-09-07:
It matches what I found. I'll try it in a moment. Thanks a lot.

closed 2019-09-07 by Neo003

gabgabe, 2020-09-02:
@Neo003 @lothean-dev  hi, did you succeed in creating Ninebot compatible BMS?

Neo003, 2020-09-08:
Sorry, I've never worked on it, I don't have enough time to try that.