BotoX/xiaomi-m365-compatible-bms #25: Issues uploading firmware

nick9one1, 2020-01-08:
I managed to flash the bootloader ok `C:\Users\Nick\Downloads\avrdude-6.3-mingw32>avrdude -patmega328p -cusbasp -U lfuse:w:0xFF:m -U hfuse:w:0xDE:m -U efuse:w:0xFD:m -U lock:w:0x3F:m -U flash:w:optiboot_atmega328_8mhz_57600bps.hex

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file "0xFF"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xFF:
avrdude: load data lfuse data from input file 0xFF:
avrdude: input file 0xFF contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified
avrdude: reading input file "0xDE"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0xDE:
avrdude: load data hfuse data from input file 0xDE:
avrdude: input file 0xDE contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xFD"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xFD:
avrdude: load data efuse data from input file 0xFD:
avrdude: input file 0xFD contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "optiboot_atmega328_8mhz_57600bps.hex"
avrdude: input file optiboot_atmega328_8mhz_57600bps.hex auto detected as Intel Hex
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against optiboot_atmega328_8mhz_57600bps.hex:
avrdude: load data flash data from input file optiboot_atmega328_8mhz_57600bps.hex:
avrdude: input file optiboot_atmega328_8mhz_57600bps.hex auto detected as Intel Hex
avrdude: input file optiboot_atmega328_8mhz_57600bps.hex contains 32768 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 32768 bytes of flash verified

avrdude: safemode: Fuses OK (E:FD, H:DE, L:FF)

avrdude done.  Thank you.

C:\Users\Nick\Downloads\avrdude-6.3-mingw32>
`

but get an error when trying to upload through platform.io

`Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Use manually specified: /dev/ttyUSB6
Uploading .pio\build\sp15s001\firmware.hex
avrdude: ser_open(): can't open device "/dev/ttyUSB6": The system cannot find the path specified.`

usbasp shows as Port_#0006.Hub_#0006 in device manager, which I think is correct for /dev/ttyUSB6?

nick9one1, 2020-01-08:
can be closed, 

wasnt using the correct platforio.ini config

platform = atmelavr
board = pro8MHzatmega328
framework = arduino
**upload_protocol = usbasp**
upload_port = /dev/ttyUSB6
upload_speed = 57600
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED

ryci-us, 2020-03-06:
So you can aswer my question - what interface you have used to upload firmware? ISP or UART ?