BotoX/xiaomi-m365-compatible-bms #26: Issues Python serial communication

TorstenE81, 2020-01-22:
Hi,
I'm currently building this BMS. Everything seems to work fine except of Python config tool. I always get the BOOTED Message but when I send debug_print() I get usually just get some hex code like 55aa0322fa0500dbfe.  If I call debug_print many times I sometimes get the desired debug information.
If I call getSettings it always ends in a Python exeption that no data was received.
I tried Python Version 3.7 on Ubuntu and 3.5 on a Raspberry. On Python 2.7 the script doesn't start.

Does anyone hav a hint? Thanks in advance 

Torsten

BotoX, 2020-01-22:
Seems like the serial UART connection is bad.
It has no error correction or even detection.
So communication errors will cause this behavior.

TorstenE81, 2020-01-23:
what's your recommendation? Can I optimize the connection or should I try
another USB UART converter?

TorstenE81, 2020-01-23:
That's the message I get after calling getSettings()

>>> getSettings()
b'55aa0322f1007079fe'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "./configtool.py", line 199, in getSettings
    d = m365_recv()
  File "./configtool.py", line 187, in m365_recv
    d = g_Queue.get(True, 1)
  File "/usr/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

TorstenE81, 2020-01-24:
> Seems like the serial UART connection is bad.
> It has no error correction or even detection.
> So communication errors will cause this behavior.

> Seems like the serial UART connection is bad.
> It has no error correction or even detection.
> So communication errors will cause this behavior.

#26
TorstenE81 opened this issue 2 days ago · 3 comments
Comments
@TorstenE81 TorstenE81 commented 2 days ago • 
Hi,
I'm currently building this BMS. Everything seems to work fine except of Python config tool. I always get the BOOTED Message but when I send debug_print() I usually just get some hex code like 55aa0322fa0500dbfe. If I call debug_print many times I sometimes get the desired debug information.
If I call getSettings it always ends in a Python exeption that no data was received.
I tried Python Version 3.7 on Ubuntu and 3.5 on a Raspberry. On Python 2.7 the script doesn't start.

Does anyone have a hint? Thanks in advance

Torsten

@BotoX BotoX commented 2 days ago
Seems like the serial UART connection is bad.
It has no error correction or even detection.
So communication errors will cause this behavior.

That's the message I get after calling getSettings()

getSettings()
b'55aa0322f1007079fe'
Traceback (most recent call last):
File "", line 1, in
File "./configtool.py", line 199, in getSettings
d = m365_recv()
File "./configtool.py", line 187, in m365_recv
d = g_Queue.get(True, 1)
File "/usr/lib/python3.7/queue.py", line 178, in get
raise Empty
_queue.Empty

Today I tried another UART USB converter. Unfortunately without success. 

Do you have some further advices?