BotoX/xiaomi-m365-compatible-bms #29: Firmware upload

kalski44, 2020-04-04:
Right now I'm just checking this great project, but one point, which is probably obvious for most of you, is not clear for me.

Do I have to use ISP programmer only to program bootloader by "AVRDUDE", and To upload firmware I should use dedicated PC interface which is delivered with BMS. Am I right?

So I should buy BMS + PC interface (there is also option, only with bluetooth interface).

BotoX, 2020-04-04:
You can program both with ISP and avrdude.
But a serial UART converter is useful for debugging and updating the firmware (not only for this device but lots of others like your internet router when flashing alternative firmware, etc.)
So I prefer to teach people how to use both.
With platformio it's just one click of a button to upload the firmware to a chip with a bootloader (okay you probably have to reset the chip/power first and then press the button with 2 seconds)
If you have any problems with the BMS you need to have a UART adapter to read out the error code etc.

kalski44, 2020-04-04:
Thank you for the answer. To assume I dont need "PC comunication module" sold with SP15SV0001-LLT.
Folowing this, I have tried to upload firmware by Serial UART adapter. Unfortunately I have some issue and I'm not sure how to solve it. Until now I made inly simple arduino projects using Arduino IDE, as a hobby.

Below step by step my actions, and results:

**1. I programmed bootloader using AVERDUE:**
```
C:\Users\laptop\Documents\PlatformIO\Projects\xiaomi-m365-compatible-bms-master>avrdude -patmega328p -cusbasp -P/dev/ttyUSB3 -U lfuse:w:0xFF:m -U hfuse:w:0xDE:m -U efuse:w:0xFD:m -U lock:w:0x3F:m -U flash:w:optiboot_atmega328_8mhz_57600bps.hex

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file "0xFF"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xFF:
avrdude: load data lfuse data from input file 0xFF:
avrdude: input file 0xFF contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified
avrdude: reading input file "0xDE"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0xDE:
avrdude: load data hfuse data from input file 0xDE:
avrdude: input file 0xDE contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xFD"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xFD:
avrdude: load data efuse data from input file 0xFD:
avrdude: input file 0xFD contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
[... 66 lines cut ...]
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Use manually specified: /dev/ttyUSB3
Uploading .pio\build\sp15s001\firmware.hex
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Users\laptop\.platformio\python37\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\laptop\.platformio\python37\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\laptop\.platformio\penv\lib\site-packages\platformio\proc.py", line 50, in run
    self.do_reading()
  File "c:\users\laptop\.platformio\penv\lib\site-packages\platformio\proc.py", line 73, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "c:\users\laptop\.platformio\penv\lib\site-packages\platformio\proc.py", line 73, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "C:\Users\laptop\.platformio\python37\lib\codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbf in position 32: invalid start byte

=========================================================== [FAILED] Took 2.26 seconds ===========================================================
```

Do you have some idea what can I do wrong?

kalski44, 2020-04-07:
It was my fault. I made the same mistake as is already solved in issue #25

closed 2020-04-07 by kalski44