BotoX/xiaomi-m365-compatible-bms #31: Low voltage

kalski44, 2020-04-12:
I connected everything according description (Battery "-" is connected to "B-" on BMS, and that's all (except plug with wires for 12 cells).
Bootloader and firmware is uploaded without errors.
Unfortunately voltage between battery + and "P-" on BMS is only ~20V and dropping to 10V in 3 seconds.
I shorted RESET with GND (as is described in trouble shots section) but it doesn't help.
What can be an issue and what else can I do?

BotoX, 2020-04-12:
Did you connect both temperature sensors?
Also debug_print is usually the answer to "why is it not working"

kalski44, 2020-04-13:
Of course I didn't connect temperature sensors... I seen that comment in "trouble shot" section, but I didn't connect it with my issue... 
I spent yesterday few hours with that issue, and solution  was so easy and obvious. Thank you for your help. 
Probably I will have some other issues (honestly speakinh I already have, but first I will try to solve myself). But I hope step by step I will finalize this project before end of quarantine ;). Thanks again.