BotoX/xiaomi-m365-compatible-bms #32: Wiring

tbone-rd, 2020-04-14:
Hi Botox

I'm unsure about wiring, because of the following warning:

> Do not cut the big - trace on the M365 ESC or it will damage the BMS because of different GND potentials on UART!

**I have no idea whats meant.. Which trace should i not cut? **

I just dont wanna cut/disconnect some obvious thing which i dont see as noob.

[image]

**Where do i connect the other wire of brake light?** 

Thank you very much

kalski44, 2020-04-14:
I'm not sure if my advice is correct, because I didn't finish this modification yet, but in my opinion your scheme is OK, just don't modify anything on the ECS PCB. I connected everything as on your sketch and nothing exploded ;). My scooter doesn't work yet, but this is rather an issue with software than hardware.

BotoX, 2020-04-14:
This schematic is totally fine, the warning was about modifying the ESC PCB.
Also your brake light is missing a 2nd wire :p that'd be P-

NetBUG, 2020-07-15:
Let me bump the topic for high voltage mods.
The circuit by @tbone-rd looks fine if the battery is below or equal to 12S (which is supported by M365 ESC).
[image]
The problem is that BMS disconnects the GROUND wire in case the battery goes below safe voltage. 

For 13S and higher voltage batteries we need another positive wire with 42V (taken from 10th cell positive terminal).
I feel it is unsafe to keep using the BMS switch in that case.

How should ESC v1.4 be wired with 13S battery? Positive wire on the ESC is supposed to be cut, so that 54V go to MOSFET upper shoulder, and 42V go to TPS and all the circuit related.