BotoX/xiaomi-m365-compatible-bms #33: Unknown (communication with ESC?) issue

kalski44, 2020-04-14:
**What have I done until now step by step:**
**1.** I uploaded modified ECS firmware with"BMS baud rate to 76800" option
**2.** Programmed bootloader by averdue (no errors)
**3.** Uploaded new BMS firmware by PlatformIO and USBASP (no errors)
 *Battery voltage is OK, RX/TX connection between ECS and new BMS is fine.

When I try to turn on scooter dashboard speaker is bipping all the time (the same result was when I tried turn on scooter with disconnected RX/TX cable before any modifications). 
I assume there is some issue with comunication between ECS and BMS...

Unfortunately I can't figure how can I check what exactly is the issue...

In description is an info about "debug_print()" option, but honestly speaking I don't know how to activate it, as I newer used python before.
I tried to use terminal and comands, as below, but as you can see error is my only result :( (evetything using this adapter: https://pl.aliexpress.com/item/2055099231.html)

```
C:\Users\laptop\Documents\PlatformIO\Projects\xiaomi-m365-compatible-bms-master_v2>python configtool.py /dev/ttyUSB3 debug_print()
Traceback (most recent call last):
  File "configtool.py", line 12, in <module>
    ser = serial.Serial(COMPORT, 76800)
  File "C:\Users\laptop\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\laptop\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\laptop\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/ttyUSB3': FileNotFoundError(2, 'System nie może odnaleźć określonej ścieżki.', None, 3)

C:\Users\laptop\Documents\PlatformIO\Projects\xiaomi-m365-compatible-bms-master_v2>
```

As I understand the main issue is COM port, but it is the same port which I used in PlatformIO to upload firmware.

tbone-rd, 2020-04-14:
I had no success to communicate with bms, but i think only connect uart was not enough, i have to finish first the cellpack.

On Windows you have to use COM_n_
To get the COM Port use this in cmd
python -m serial.tools.list_ports -v
Then you can add it as parameter or just edit your configtool.py for COM3:
COMPORT = sys.argv[1] if len(sys.argv) > 1 else 'COM3'

Lets see if we can finish this project soon:-)

kalski44, 2020-04-14:
Thanks for advice. Unfortunately it doesn't change anything :(.
Tomorrow I will try to install Linux on virtual machine. I don't want to give up when I have made all hardware already.

I hope you won't have any issues. If you will be able to comunicate with BMS please let me know here or by email. (doesn't matter if it will be tommorrow or in two weeks) If you would be so kind I would like to check with you step by step what did you do and what I'm doing wrong.
Of course If you will have some other issues or doubts don't hasitate to contact with me... apparently we are on the same path, in the same time, so we can support eachother ;)

*my email: karol.t144@gmail.com

BotoX, 2020-04-14:
you can also try any other terminal emulator program and send this in HEX: 55aa0322fa0500dbfe
also try sending it multiple times, UART is a shit protocol with no error correction
and do you get BOOTED! when you reset the BMS while listening to serial?

kalski44, 2020-04-15:
Now I connected serial to UART adapter (https://pl.aliexpress.com/item/32717057832.html) to listen serial port and I used serial monitor from Arduino IDE. 
BMS should send "BOOTED!" each time after reset (RST to GND shortage), am I right?
Unfortunately I don't have any message (I tested all available serial port speeds and I used both possible RX/TX cables configuration) :(

BotoX, 2020-04-15:
> BMS should send "BOOTED!" each time after reset (RST to GND shortage), am I right?

Yes.

> Unfortunately I don't have any message (I tested all available serial port speeds and I used both possible RX/TX cables configuration) :(

I'd say that you don't have any firmware on your BMS.  
Are you sure you flashed the bootloader AND the firmware, maybe you only flashed the bootloader?

tbone-rd, 2020-04-15:
@BotoX 
1.) Does the BMS send "Booted" when batterypack not connected? (I have not finished yet)
Only UART Adapter (3.3V or 5V or doesnt matter?) connected to Computer.
2.) Does configtool.py work when batterypack not connected only UART?

Then i can try here too, till i have finished cellpack

kalski44, 2020-04-15:
Hmmm.. This is the message from PlatformIO when I used "upload button":
There are some warnings, but at the end it seams to be fine.

```
Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing sp15s001 (platform: atmelavr; board: pro8MHzatmega328; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/pro8MHzatmega328.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Pro or Pro Mini ATmega328 (3.3V, 8 MHz)
HARDWARE: ATMEGA328P 8MHz, 2KB RAM, 30KB Flash
PACKAGES: 
 - framework-arduino-avr 5.0.0
 - tool-avrdude 1.60300.190424 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PinChangeInterrupt> 1.2.6
|-- <SoftI2CMaster> 1.0.0 #a60c4da
|-- <EEPROM> 2.0
Building in release mode
Checking size .pio\build\sp15s001\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  54.2% (used 1111 bytes from 2048 bytes)
Flash: [======    ]  62.9% (used 19330 bytes from 30720 bytes)
Configuring upload protocol...
AVAILABLE: usbasp
CURRENT: upload_protocol = usbasp
Looking for upload port...
Uploading .pio\build\sp15s001\firmware.hex

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file ".pio\build\sp15s001\firmware.hex"
avrdude: writing flash (19330 bytes):

Writing | ################################################## | 100% 11.42s

avrdude: 19330 bytes of flash written
avrdude: verifying flash memory against .pio\build\sp15s001\firmware.hex:
avrdude: load data flash data from input file .pio\build\sp15s001\firmware.hex:
avrdude: input file .pio\build\sp15s001\firmware.hex contains 19330 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 9.97s

avrdude: verifying ...
avrdude: 19330 bytes of flash verified

avrdude: safemode: Fuses OK (E:FD, H:DE, L:FF)

avrdude done.  Thank you.

============================================================== [SUCCESS] Took 23.98 seconds ==============================================================
```

BotoX, 2020-04-15:
You have to power the BMS by connecting the balance connector, only like that will the TI chip work properly.
If you only want to test the ATMega firmware you just wrote with platformio you can hook it up to GND and +5V over ISP, but don't connect anything else.
I'd suggest you to finish your battery first and then mess with the BMS, there's no point to the BMS without a battery :p

tbone-rd, 2020-04-15:
@kalski44 

i got this warnings too, so far i have read its not critical.
sck period is because of old firmware on usbasp

I got exactly same output instead of fuses
avrdude: safemode: Fuses OK (E:FD, H:DE, L:E2)
I'm on bms V1.5 (yeah, my stuff is chilling around for years) and your on V1.6?

@BotoX 

battery will take some days..
Only GND and +5V to ISP. But also only TX and RX to UART (Signal Level 5V)?
Then i should get a "booted" and can also modify settings via configtool.py via UART

kalski44, 2020-04-15:
@tbone-rd 

Yes I have BMS 1.6, that's why different fuses

BotoX, 2020-04-15:
> @kalski44 
> battery will take some days..
> Only GND and +5V to ISP. But also only TX and RX to UART (Signal Level 5V)?
> Then i should get a "booted" and can also modify settings via configtool.py via UART

Yeah that should work, but I wouldn't worry about it until your battery is finished.

kalski44, 2020-04-15:
I have BOOTED! message.
The issue was so stupid that I shouldn't even admit that, but I mixed pins and connected GND (from ISP programmer) to VCC on BMS instead of GND...

[image]

Now I see "Boot" message, but still I don't know how to run "debug_print()" command :(.
I also tried to send "55aa0322fa0500dbfe" by serial monitor terminal, few times one, by one, but without any reply from BMS.

kalski44, 2020-04-15:
I have some progress... 
When I use commands in terminal:
`python -i configtool.py COM8` , and after that `debug_print()` and `getSettings()`
I have some reply (different than serial port issue). Unfortunately it doesn't tell me anything :(. Does it mean something for you?

```
C:\Users\laptop\Documents\PlatformIO\Projects\xiaomi-m365-compatible-bms-master>python -i configtool.py COM8
>>> debug_print()
b'55aa0322fa0500dbfe'
>>> getSettings()
b'55aa0322f1007079fe'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "configtool.py", line 211, in getSettings
    d = m365_recv()
  File "configtool.py", line 199, in m365_recv
    d = g_Queue.get(True, 1)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.752.0_x64__qbz5n2kfra8p0\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty
>>>
```

BotoX, 2020-04-15:
try it multiple times, as said serial is pretty unreliable.

kalski44, 2020-04-15:
I tried 20 times, but I'm wandering about port speed. In monitor options I can choose only speeds as below:
[image]

but in "main.cpp" file, speed is defined as 76800. This is necessary to communicate with ESC as I understand, but is it OK for communication with serial monitor?

[image]

BotoX, 2020-04-15:
It definitely needs to be 76800 baud.
And it's a HEX string, not ASCII.
Better use the python configtool

kalski44, 2020-04-16:
Thanks. 
In Python tool and after multiple (close to 20) tries I have output details from BMS.

I don't know how to interpret some of the values, but everything seams to be fine. 
Unfortunately it means that I have no clue what can be an issue with my scooter...

I uploaded firmware with 76800 baud rate. There was no errors, but when I connect new BMS I have only bipping from dashboard :(.

```
>>> x00 SYS_STAT:  00000000
0x01 CELLBAL1:  00000000
0x04 SYS_CTRL1: 00011000
0x05 SYS_CTRL2: 01000011
0x06 PROTECT1:  00011101
0x07 PROTECT2:  01111001
0x08 PROTECT3   00010000
0x09 OV_TRIP:   10110000
0x0A _TRIP:   11001001
0x0B CC_CFG:    00011001
0x32 CC_HI_LO:  0
0x2A BAT_HI_LO: 27194
ADCGAIN:        377
ADCOFFSET:      45
ER ERROR:     0

uptime: 48426
Battery voltage: 41550 (27195)
Battery current: 0 (0)
SOC: 99.99
Temperature: 20.60 20.20
Balancing status: 0
Cell voltages (15):
3463 (9067), 3455 (9047), 3469 (9083), 45 (0), 3458 (9054), 3467 (9079), 3473 (9093), 3480 (9114), 45 (0), 3464 (9070), 3478 (9107), 3472 (9091), 3422 (8958), 45 (0), 3444 (9017)
Cell V: Min: 3422 | Avg: 3462 | Max: 3480 | Delta: 58
maxVoltage: 4157
maxDischargeCurrent: 1
maxChargeCurrent: 0
XREADY errors: 0
ALERT errors: 0
P errors: 0
OVP errors: 0
SCD errors: 0
OCD errors: 0

DISCHG TEMP errors: 0
CHG TEMP errors: 0
CHG OCD errors: 0
```

BotoX, 2020-04-16:
getSettings() populates g_Settings (unless you see an exception)

You could try swapping RX/TX on your scooter / BMS.
Which error is your scooter beeping? I am guessing 21 - Battery communication error ?

kalski44, 2020-04-16:
I have small success. 
Apparently main issue was with m365 firmware, because when I recovered ESC firmware using ST-link, and uploaded modified firmware again I can see (in "M365_downG") battery status, with correct voltage for all 12 cells, unfortunately not everything works fine... I tried with firmware for m365 and m365pro, as below:

**Firmware m365** (link from this project description) with standard dashboard:
BotoX settings + 
-max speed increased to 35
-BMS baud rate to 76800
-Wheel speed multiplier modified (I have 10 inch wheels)

RESULTS: 
- battery status on dashboard is max (but 12s battery voltage is only 41.5V)
- When I start scooter (even if it runs only 1s) it doesn't stop without using brake.
- "m365 Dashboard" app doesn't show any speed (alvays 0)

**Firmware m365pro** (https://hackm365.com/pro/  "M365 Pro Fases") with m365pro dashboard.
Most of the settings I used the same as in previous firmware

RESULTS: 
- Speed is shown correctly on the dashboard and in the app
- Scooter runs as should (only when I use throttle), but 
- Cruise control doesn't work at all :(
- Battery status on dashboard is max (but 12s battery voltage is only 41.5V) :(

BotoX, 2020-04-16:
Well what's your actual battery voltage measured with a multimeter?
The BMS shows 100% capacity when you reset it, you have to charge it full then.

kalski44, 2020-04-16:
Thnks for that info. I didn't charge it yet. I have measured voltage 41.5V.

One (I hope the last one) question:
You mentioned about charger which can give up to 5A on the output.
I have currently charger with appropriate voltage, but only 2A output. Can I use it without BMS firmware modifications? 
*I know charging time will be long with 2A only, but this isn't an issue for me.

BotoX, 2020-04-16:
Raw value:
Battery current: 3021 (358)
358

The BMS only limits the maximum charge current, the actual current is always set by your charger.
A BMS is just a very smart on/off switch and balancer.