BotoX/xiaomi-m365-compatible-bms #34: Very slow charging

tbone-rd, 2020-05-15:
@Botox

First of all, thank you very much for what you have done for the M365 Community! Its really outstanding! M365 is ok, with your custom Firmware it is good (was awsome before:-)), with Battery/BMS Mod, it is awsome!

As you can guess, i have finished the battery upgrade finally and did a test ride, it was like nigth and day! The M(365) stands for Mountain, now. I can climb/accelerate the steepest hills with 20km/h and i can drive about 40km careless.
What i'm telling you... you allready know it:-) Thank you, now i have the scooter i wanted.
I'm working now on the final little piece (temp alarm) to not fry the motor, because it gets very hot. If somebody would find that useful too, i can post it.

Charging Problem:

ELG-240H-48A => 49.2V / 4A / for End Voltage = 4.1V

I have noticed that charging at the end needs very much time.
Endless from 4.0-4.1V

Minutes / Charging Current / Cell Voltage
  0m / 3.55A / 3.94V (47.33V / 86% / 11354mah)
  5m / 3.20A / 4.00V
13m / 2.40A / 4.02V
19m / 1.85A 
29m / 1.30A / 4.06V
34m / 1.12A
38m / 1.00A / 4.07V
53m / 0.80A
60m / 0.70A / 4.08V ( 48.96V / 98% / 12967mah)

1h for add 0.14V, and still far far away from 4.1V

I understand that at full capacity the current has to be lowered at the end.
But because we are going only to 4.1V, could we not just rush there?

In several diagrams found transition from CC to CV is normaly at 4.1V
Can we change the transition voltage? i have not found in code.
Is it possible, that in the code is assumed,
that the custom config value (4.1V) is taken as full capacity instead of 4.2V into charging curve and this will make cc/cv too early? 

Shunt not calibrated, but looked nearly same as on my cheap multimeter.

Can i do something with this parameters?

> // setBalancingThresholds 
> uint16_t balance_minIdleTime = 1800; // s 

what is this for?

> uint16_t balance_minVoltage = 3600; // mV 

would it not be better to set higher (4.00V) for quicker charging before balancing?

> // setIdleCurrentThreshold 
> uint16_t idle_currentThres = 500; // mA

I dont understand this value. Is this the current for CV? Maybe we can set this to 4A assumed when only charging to 4.1V ?

Thank you

BotoX, 2020-05-16:
This has nothing to do with the BMS, you will get the same behavior when charging without a BMS.
All the BMS can do is disconnect your battery if there is something wrong.
(It can also discharge cell pairs over the balance pairs to bring them closer to the other cells.)

Charging is done by the charger, none of the BMS variables can influence it.

When all the cell voltages are above balance_minVoltage, then balance_minIdleTime is how many seconds the abs(current) has to be below idle_currentThres to start the balancing process.

[image]

What you are are observing is the CV phase of your charger.
Once the current goes below your set max current (CC current on your charger) it goes into CV mode. That means it just sets the max voltage and the current sets itself depending on the cell voltage and internal resistance.

tbone-rd, 2020-05-16:
thanks, i was thinking the bms will reduce current.

Do you have the same charger, rigth?
Same slow charging so early?

About internal resistance:
New cells samsung 35E, same batch. Delta 5mV
Nickelstrips 0.2/8mm
12AWG

Its because i have used the orginal chargingport with the thin cable
and having to much resistance, maybe?

Can i somehow measure the internal resistance, to see if something can be fixed?

> Keep in mind that the chager output voltage has to be adjustable to 49.2V (50.4V if you plan on running 4.2V per cell).

I have adjusted from 4150 to 4100 because it doesnt look matching (12x 4150 = 49.8V)
What does this change, does my battery just shows quicker and longer full than yours?
full_voltage = 4000; // mV your default 4150

The Vout / Iout Potis on Charger has to adjusted without load, correct?
Or is it better to adjust the max Current when in CC mode?

Is this an idea, makes it any sense?
Change Voltage on Charger only to 50.4V (for 4.2V) then the bms will balance the cells constantly
down by full_voltage = 4100; // mV OR  a i have to set OVP_voltage = 4100; // mV ?
But balancing can only handle 50-60mA, so will they go above 4.1V anyway?