BotoX/xiaomi-m365-compatible-bms #36: 1% battery, and scooter engine off with still 3,6V cell

kalski44, 2020-06-24:
Did someone meet such issue?
My scooter with new BMS is discharging really fast (range is 10km), despite that I have two additional battery packs (12s3p instead olyginal 10s3p). I thought it is due to higher current settings, but today I discovered that scooter is showing "battery empty", and don't want to run engine, when in battery details battery cells have still nominal 3,6V, and total battery voltage is 43V. I suppose it should be still possible to drive with such voltage... Do you have Idea where can be an issue?
I used default settings in "main.h" file for BMS firmware.

BotoX, 2020-06-24:
You should probably adjust your battery capacity then...
This BMS firmware uses coulomb counting for capacity tracking. Which means it integrates/adds up/counts the current which is flowing through the BMS to subtract from the capacity you told it.
The stock BMS uses voltage.

kalski44, 2020-07-05:
Thank you for advice. I used 12400mah as battery capacity and now it works fine.
Apparently with 12s this value must be set, even if I'm using only 3 cells (7800mah) battery packs.

BotoX, 2020-07-06:
Your shunt resistor value could also be wrong.
You can calibrate it using a known current through the BMS (charger with X Ampere for example or a programmable load with X Ampere)
Then calculate the shunt resistor value by R = U / I
U being the voltage drop on the shunt resistor (can read this out from debug_print raw value * 8440)
And I = your known current