BotoX/xiaomi-m365-compatible-bms #37: Why the 12s is the limit for the controller?

cptX, 2020-06-26:
Hi, why the 12s is the limit for this controller? From the datasheet of the TPS54160 the absolute maximum VIN is 65V. 12s maximum voltage is 50.4V. I understand that braking increases the VIN but I don't know how much. Does it increase it more than 15V? Why a 13s battery that has a maximum voltage of 54.6V will burn the controller? Aren't 10V voltage margin enough?
Have you tried it in real life and saw that it burns the controller?

BotoX, 2020-06-26:
Because of regen braking.
And yes a few people have tried and destroyed their controllers.
There are modifications which can be done to fix this.
Or the newer V2.1 ESC has a higher voltage DC converter, just need to replace the main capacitor which is 63V max.

cptX, 2020-06-26:
OK so the 12s is guaranteed? What about the pro? The 12s is guaranteed for the pro too, where the regen braking is more powerfull and sometimes burns mosfets?