BotoX/xiaomi-m365-compatible-bms #39: Charging improvements

NetBUG, 2020-07-20:
Hi,
I see several things that could be improved with charging.

1) Right now charging FET is enabled always when the BMS is not in protection mode. That may lead to fast charger port oxidation.
Is there any voltage divider on the charger input? It would be better to keep charger FET closed unless there is some voltage supplied

2) The overvoltage protection does not shut down automatically, the BMS remains locked until reboot (which implies disassembling the scooter). The Xiaomi BMS had a RC circuit that triggered BMS reset when charger is plugged in. Should that be implemented? Or it's better to set timeouts for OVP until the charging voltage is not removed?

3) Is it possible to implement CC mode using the charger MOSFET, or its driver is too slow for that?
I still don't have a good charger for 13S yet, and think it could be great to use any voltage source of nearly 60V (without the need of CC circuit) to run it

BotoX, 2020-07-20:
1) Mosfets are asymmetrical switches, they can only control current in one direction.
Charging FET can only block charge current, you can still discharge from charge port when the charge FET is off (and discharge FET is on). The current flows through the body diode in reverse mode.
2) I'll have to check that. But OVP should only disable charging and reset automatically. Discharging should still work.
3) The TI IC can only turn the FET fully on or off, also using a FET as a resistor to burn off the difference in voltage sounds like a bad idea.

NetBUG, 2020-07-21:
Thank you for reply!
1) Got it! I feel when the FET is closed, there is much lower voltage and the current is close to zero. However, I won't insist that charging FET needs to be kept closed all the time unless charging is in progress, and just will add an "ideal diode" between charging port and the battery (as seen in https://www.davidpilling.com/wiki/index.php/IdealDiode, http://jiggerjuice.info/electronics/projects/power/ideal-diode.html)
2) It does not even after two days. I had full charge (one cell reached 4.18V due to imbalance) on Friday, had no rides on the weekend, yesterday came to the lab and tried to charge -- the controller was locked. I had to disconnect balancing port to reset it. Right after that charging resumed
3) Yep, using the transistor as active load is a terrible idea :) I just thought the IC can allow to implement something like PWM mode which can be controlled programmatically after measuring voltages and currents. In fact, CC chargers do exactly the same with the same MOSFETs inside and some simple LC filters.
I think I can check it on my own tomorrow

closed 2020-07-23 by BotoX

BotoX, 2020-07-23:
You were right about the reset not working, should be fixed now.
I've introduced an error there while refactoring the code and copy&pasting too much, thanks.

commit 85d8d2cc (2020-07-23): Fix overvoltage error not clearing, copy&paste oops. Fixes #39