BotoX/xiaomi-m365-compatible-bms #41: Is it possible to do smart bms with more 30A ? For example 40 or 60A ?

g4seb, 2020-08-28:
Hi,
i search a bms for my custom battery. But if i understand, yours is a 30A. Is it possible to do this mod with a more amper bms ?
Last question, some people sell bms or not ?
Regards

BotoX, 2020-08-29:
The 30A version has 4 or 5 of these Mosfets in parallel: https://lcsc.com/product-detail/MOSFET_KIA-Semicon-Tech-KNB3308A_C176861.html
I did measure them to have around 6-7 mOhm when driven at 12V by the BMS.
So that's like 1.4 mOhm total.
At 50A it'd be this much waste heat: P = R * I^2 = 1.4m * 50*50 = 1.4m * 2500 = 3500mW = 3.5W
That's no problem for the BMS

g4seb, 2020-10-01:
Sorry for delay. Hi, ok, so i try to this with my BMS with 60A : https://fr.aliexpress.com/item/32819508078.html?spm=a2g0s.9042311.0.0.79106c376b8ChE.
I arrive to patch this, apparently, all is ok, i'm doing a cfw with 76800 bauds, but when i put it on my scooter, nothing append. It doesn't communicate with scooter. Do you have a idea to check that ?

g4seb, 2020-10-01:
Hi have this message on m365_downG 

[image]

BotoX, 2020-10-01:
If you've only flashed the bootloader then you also have to flags the actual BMS firmware.
Either through ISP or over serial UART with platformio.
Read the README.md

g4seb, 2020-10-02:
Hi, shit, i don't flsh FW. I think it's the reason why i have BMS version to 0...
I have a ips, and i buy a serial uart. But on then readme, i doesn't see where did i put the cable on bms with serial UART  ? There are just picture with IPS.
Sorry for my question, but i really want to understand.