BotoX/xiaomi-m365-compatible-bms #42: Error "Cant open device /dev/ttyUSB0

Hazuka, 2020-08-28:
all is in the title. I have the USBasp and the V1.6 BMS and i CANT find any solutions to program it.. Any help? :)

Hazuka, 2020-08-28:
[image]

i have no com port in Device list of PlatformIO... any advice?

Hazuka, 2020-08-28:
[image]

But its detected by Windows. what goes wrong ? XD

BotoX, 2020-08-29:
Upload protocol Arduino is not usbasp
It's for uploading the code via uart when using a bootloader
Usbasp is used with avrdude to upload bootloader
But it can also be used for firmware with avrdude
Or with platforming, read the documentation of platformio avr

Hazuka, 2020-08-30:
> Upload protocol Arduino is not usbasp
> It's for uploading the code via uart when using a bootloader
> Usbasp is used with avrdude to upload bootloader
> But it can also be used for firmware with avrdude
> Or with platforming, read the documentation of platformio avr

i cant see any port.. can you explain what need to do?

Hazuka, 2020-08-30:
Finally need to removed ENTIERELY the WinUSB drivers for it to make an exchange for libusb drivers. problem solved. Thanks you Botox for your message ;)

> Upload protocol Arduino is not usbasp
> It's for uploading the code via uart when using a bootloader
> Usbasp is used with avrdude to upload bootloader
> But it can also be used for firmware with avrdude
> Or with platforming, read the documentation of platformio avr

closed 2020-08-30 by Hazuka