BotoX/xiaomi-m365-compatible-bms #43: Alternatives for JBD-SP15S001?

IuliuNovac, 2020-10-21:
Hi, since the new version the supplier has (v1.6) dosn't have the ISP pins, has anyone tried an diffrent board?
P.s: I destroied one JBD-SP15S001 v1.6 already, by tring to solder.

clemtrinc, 2021-01-17:
Hi, did you ever find a solution ?
I just finished assembling the pack and discovered my BMS is v1.7 and no ISP connection available...

IuliuNovac, 2021-01-17:
Yes, i found an emulator. So I have an bms and an emulator to fool m365 that i have the right bms.

MBryde, 2021-01-17:
@IuliuNovac What emulator did you find?

IuliuNovac, 2021-01-17:
I don't remember, i know i used this diagram from this one.
https://github.com/CamiAlfa/M365_BMS_EMULATOR

clemtrinc, 2021-01-17:
Yes, that's what I will do probably, to bad you then loose the battery level on screen.
What a shame that they changed the bms...