BotoX/xiaomi-m365-compatible-bms #50: Problem collect2.exe: error: ld returned 1 exit status

Adri-Dax, 2021-03-07:
Goodnight. I am trying to compile the software for the botox bms. I always get this error even if I don't modify any data and I try to get this error. Because it can be? Thanks a lot[https://gyazo.com/b96deac829e8d47a71749b10672722d6](url)
Compiling .pio\build\sp15s001\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\sp15s001\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\sp15s001\FrameworkArduino\wiring_pulse.S.o
Compiling .pio\build\sp15s001\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\sp15s001\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\sp15s001\libFrameworkArduino.a
Linking .pio\build\sp15s001\firmware.elf
C:\Users\Adri\AppData\Local\Temp\ccviLg1E.ltrans0.ltrans.o: In function `_Li2c_read_bit':
<artificial>:(.text+0xe6c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
<artificial>:(.text+0xe70): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
<artificial>:(.text+0xe7c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
C:\Users\Adri\AppData\Local\Temp\ccviLg1E.ltrans0.ltrans.o: In function `_Li2c_put_ack_high':
<artificial>:(.text+0xe9c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
<artificial>:(.text+0xea8): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
<artificial>:(.text+0xeaa): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
C:\Users\Adri\AppData\Local\Temp\ccviLg1E.ltrans0.ltrans.o: In function `_Li2c_write_high':
<artificial>:(.text+0xeda): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
<artificial>:(.text+0xee8): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
C:\Users\Adri\AppData\Local\Temp\ccviLg1E.ltrans0.ltrans.o: In function `_Ldelay_scl_high':
<artificial>:(.text+0xef0): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
C:\Users\Adri\AppData\Local\Temp\ccviLg1E.ltrans0.ltrans.o: In function `_Li2c_get_ack':
<artificial>:(.text+0xefc): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
C:\Users\Adri\AppData\Local\Temp\ccviLg1E.ltrans0.ltrans.o: In function `_Li2c_ack_wait':
<artificial>:(.text+0xf0a): additional relocation overflows omitted from the output
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\sp15s001\firmware.elf] Error 1
========================== [FAILED] Took 4.11 seconds ==========================

mottscanley, 2021-03-31:
add some older versions of framework-arduino-avr and toolchain-atmelavr to platform_packages in .ini file

BotoX, 2021-03-31:
Seems to be an issue with https://github.com/felias-fogg/SoftI2CMaster and Atmel AVR 3.3 (worked on 2.2)
I've found some forum post where they've replaced all `rjmp` with `jmp` and `rcall` with `call` in SoftI2CMaster.h
Done that and it compiles for me now.
See: https://github.com/felias-fogg/SoftI2CMaster/issues/50

I've pushed a commit which should fix that for you.

commit 90cac0a2 (2021-03-31): Fork SoftI2CMaster to fix link error. Fixes #50
closed 2021-03-31 by BotoX