BotoX/xiaomi-m365-compatible-bms #55: Recover from cut the trace (Yes I am stupic I didnt read about it ~2 years ago or this warning was not there :-))

AdamWeglarz, 2021-09-25:
Hi,

I had mine setup with cut the trace working for more than a year. Later I stopped using my scooter due to covid. Now I wanted to check everything and make it right so I have removed cut the trace from my esc and now I have problem with balancing 11th of 13 cells. It stays 1,5V below other when fully charged (4,05) and other are around 4,19.
Is there any way to fix BMS?

BotoX, 2021-09-25:
Let the BMS sit for a few days/weeks and it should balance out, slowly because 1.5V is a loooot, that cell group might be damaged and needs to be replaced
Or if your firmware is old maybe there is bug in it, update could help.
However I'd personally connect a single cell charger to that cell group and charge it myself.
And check with a multimeter before it the BMS is measuring correctly
Post your debug_print() in any case

AdamWeglarz, 2021-09-25:
Hi,

I am recharging this cell. And than it is ok. But after night BMS is decharching this cell even without changer on fully charger battery! This is one issue. Second issue is that when I charge battery, this cell remains at level of 4 maybe 4,05 volt and rest is around 4.2. 
I have exchanged also batteries to checked ones with good capacity (although old ones are also good).
I did update firmware succesfully, but it was latest one previosly as I already make an upgrade as I had issues with previous one which get stucked some time ago.

I am not able to get debug. I have problem with sending commands to controler with my serial port. I am able to read BOOTED, but neither with your script neither with putty I cannot send anything. Any hint how what can be done wrong (I am able to connect only on windows using FTDI usb to serial - most reliable one I know)

BotoX, 2021-09-26:
Unplug the BMS from the battery (big balance connector) and let it sit for a day, check if the cell is discharging.
Is the voltage reading on that cell group correct in the m365 app?

AdamWeglarz, 2021-09-26:
OK, I have balanced manually all cells. 

Should I leave ESC connected?

Adam

BotoX, 2021-09-26:
I thought you removed the battery from your scooter lol
ESC can stay connected, doesn't draw and power when it's off.
Take note of the cell voltages and then check again in 24 hours

AdamWeglarz, 2021-11-06:
Hi,

After one day when BMS was connected almost 30% of battery was off and cells 9-13 were significantly more discharged.
This is way. One resistor with marking 151 and one probably transistor marked y2 were getting hot.
Do you have an idea what exactly is this Y2 element? I have written to LLC - I hope they will answer, but if not...
[image]
a06bb954dd75.jpg)
[image]
[image]
[image]

BotoX, 2021-11-06:
The three legged one is a transistor that can be switched on and off by the TI BQ chip through I2C/software.
When it turns on that cell is discharged through the resistors on the other side.
That should only happen when my software detects (from the voltage readings of the TI BQ chip) that one/some cells have a higher voltage than the others.
As always, I'd recommend a debug_print() - you can see the voltages there and which cells are currently being balanced/discharged.
However if the battery is already unbalanced then obviously the BMS is going to try and balance it.
So if you suspect a hardware issue I'd suggest manually balancing those cells (to a certain degree)
and then setting the balance threshold in your config very high:
https://github.com/BotoX/xiaomi-m365-compatible-bms/blob/master/src/main.h#L71
Like to 500mV, then the software would not turn on balancing and you can check if the BMS is still discharging the cell.
Could be a faulty transistor, or faulty voltage reading which triggers balancing wrongfully, etc.

AdamWeglarz, 2021-11-07:
I will try to replace this transistor and maybe resistor. Thats why I am looking for what it is.
I have bought second BMS and this second one is doing great job balancing battery after this faulty one.

Adam

AdamWeglarz, 2021-11-08:
I have measured this transistor with diode test and it it short circuited between base and collector.
Based on marking and parameters I think this might this one SS8550
[ss8550_galaxy.pdf](https://github.com/BotoX/xiaomi-m365-compatible-bms/files/7499554/ss8550_galaxy.pdf)

They are available in Poland in one shop - I keep finger crossed. Resistor keeps proper parameters

AdamWeglarz, 2021-11-10:
I was able to upload new FW
[image]
First check:
[image]
after one hour:
[image]
started to charge battery:
[image]

It a great chance that I have fixed it successfully!

Adam

AdamWeglarz, 2021-11-11:
Its charged fully and perfectly balanced. Tomorrow is time for drive test :-)

AdamWeglarz, 2022-01-17:
BMS is totally working!

closed 2022-01-17 by AdamWeglarz