BotoX/xiaomi-m365-compatible-bms #58: SP15S051 v1.0 BMS?

hoboct, 2022-05-25:
I recently ordered a 12S BMS from the listed seller and I got what looks to be a new version: JBD-SP15S051 v1.0

Photos are here: 

https://ibb.co/sQhVM9f - board, MCU side
https://ibb.co/25TQw9x - board, FET side
https://ibb.co/BT0fLcG - attempted cllose up on MCU

Does anyone know if this will work with this firmware?

It has a Nuvoton MCU, I can't make out the exact number as there is a clear epoxy over it.

There are also 5 through hole pins VREG/DATA/CLK/RST/GND, I'm guessing this is I2C and not usable to reprogram the firmware (i.e. i'll still have to jumper pins on the MCU)

BotoX, 2022-05-25:
If the MCU changed from ATMEL to something else then this firmware will need to be rewritten in order to work.
They've used conformal coating in the past, you should be able to scratch it off with your fingernails.

zoombahh, 2022-05-31:
I Ordered a 10s 30A BMS for another project and got the JBD-SP15S051 V1.0 too.
Its 15s pcb with only 10 lines populated and a sop-30 on a sop-48 footprint to interface with the cells.

LLT confirmed the change by email when I asked about the product page (same link, new pictures, only 2port) and said they only have the "updated Version" without a P- Port  and no remaining stock of the old model.

The MCU seems to be a nuvoton NANO 100SD3BN, an Arm Cortex-M0 with 64k of flash :/

got some overview and macro shots:
MCU side
[image] [image] [image]

Mosfet side.
[image] [image] [image] [image] [image]