BotoX/xiaomi-m365-compatible-bms #60: Arduino bms

Zanooon97, 2022-08-04:
Hi, can you make an arduino ini where i set the values, for example percent left, voltage, current, battery temp.
I am not able to do that, and you know already the bms protocol i think. Only a simple code that imitade the original xiaomi bms.

BotoX, 2022-08-05:
https://github.com/CamiAlfa/M365_BMS_EMULATOR

Zanooon97, 2022-08-05:
Sir Botox, i swear this is not working for me...
Flashed the new bootloader so watchdog works.
Flashed the code but still no bms data.
Downgraded already to first fw, upgraded step by step to 1.7 and still not working.
I am able to read my bms on rx tx so my arduino works!
My be becaus the newer board. i see he is using the 1.4 board, i am using the 2.1.
Bit i think the firmware is the same.
Also changing the code a little bit, evitati g the error from under voltage or overvoltage be send still not able to read some data.
I think is usefull to have a right diagramm on the dashboard app.