BotoX/xiaomi-m365-compatible-bms #61: Development of Bafang UART BMS

iamniels, 2023-01-15:
I read your request here https://endless-sphere.com/forums/viewtopic.php?t=99137 and was wondering whether you are still interested? 2nd hand market places here in Europe are filled with "defect" Bafang batteries, because the BMS gives up, while (most of) the cells are fine. 

I'm an electrical engineer, but it would be my first time reverse engineering a foreign protocol, so I can use some help :-)

BotoX, 2023-01-16:
The protocol should be easy enough that I can reverse engineer / implement it in this BMS firmware.
I've already written a parser/proxy for the bafang protcol on esp32, to implement on the fly programming.
Code here: https://git.botox.bz/BotoX/bafang-esp32

If you have access to bikes/batteries that talk this bafang battery protocol I'd love to get a communication dump.
Serial converters and logic analyzers are still cheap and plenty from aliexpress.

fyi. Serial on my BBSHD/DPC18 is using 1200 baud 8N1

iamniels, 2023-01-24:
9600 baud 8N2. See the attached file. I have called one line BikeTX, because BatteryTX comes shortly after BikeTX.
[Packet.txt](https://github.com/BotoX/xiaomi-m365-compatible-bms/files/10494467/Packet.txt)

Is this enough?

BotoX, 2023-01-25:
```
[0x3A, 0x16, 0x0F, 0x01, 0x00, 0x26, 0x00, 0x0D, 0x0A]
[0x3A, 0x16, 0x0F, 0x04, 0xF5, 0x12, 0x00, 0x00, 0x30, 0x01, 0x0D, 0x0A]

[0x3A, 0x16, 0x09, 0x01, 0x00, 0x20, 0x00, 0x0D, 0x0A]
[0x3A, 0x16, 0x09, 0x02, 0x39, 0xA9, 0x03, 0x01, 0x0D, 0x0A]

[0x3A, 0x16, 0x0A, 0x01, 0x00, 0x21, 0x00, 0x0D, 0x0A]
[0x3A, 0x16, 0x0A, 0x02, 0x00, 0x00, 0x22, 0x00, 0x0D, 0x0A]
```

This is definitely not part of the Bafang BBS protocol, looks like Kingmeter 901U to me:
https://github.com/jenkie/Arduino-Pedelec-Controller/blob/master/Arduino_Pedelec_Controller/display_kingmeter.cpp#L268

https://github.com/EBiCS/EBiCS_Firmware/blob/master/Src/display_kingmeter.c#L351

Or this "NCM Venice" / "C7 Display"
https://www.pedelecforum.de/forum/index.php?threads/kommunikation-zwischen-c7-display-und-motorkontroller-ncm-venice-das-kit.57050/
https://www.pedelecforum.de/forum/index.php?threads/kommunikation-zwischen-c7-display-und-motorkontroller-ncm-venice-das-kit.57050/page-11#post-1710403

iamniels, 2023-01-25:
Thanks for those links! it is a Bafang M420 controller.

closed 2023-01-26 by iamniels